sinara-hw/FMC_Shuttler #1: WR clocking

gkasprow, 2018-12-04:
@hartytp shall we add WR clock recovery to Shuttler as well ? I think we should.

hartytp, 2018-12-04:
@gkasprow I that for Shuttler we will definitely want to use a cleaned recovered clock -- an external SMA clock input should not be required.

AFAICS, the Si5324 performance is fine for Shuttler (100ps drifts in the clock phase should be fine for this design), so it's not important whether we use that or WR.

However, IMO, if we're going to move to WR for Sinara generally then we should include that here as well.

gkasprow, 2018-12-04:
I think we should unify clocking and CDR in all designs and include WR as well.

gkasprow, 2020-07-16:
not relevant anymore since all WR clocking is installed on the carrier board

closed 2020-07-16 by gkasprow