sinara-hw/FMC_Shuttler #2: [RFC] New AMC: Shuttler (high-speed multi-channel DAC)

marmeladapk, 2018-12-04:
_From @hartytp on [2017-08-16 05:24](https://github.com/sinara-hw/sinara/issues/253)_

I've started sketching out requirements for a fast DAC to be used for ion shuttling/splitting. It's on the [Wiki](https://github.com/m-labs/sinara/wiki/Shuttler).

All comments welcome...

tprzywoz, 2018-12-05:
Schematics are nearly done so question about AFE: what was the conclusion?
I noted only amplifier parameters: gain 3, supply +/- 2.5 V to give +/- 1.5 V at the output. Is it actual? What are the filter requirements?

dhslichter, 2018-12-05:
@tprzywoz answered in a new issue (this one is so bloated and long): https://github.com/sinara-hw/Shuttler/issues/3#issue-387849998

gkasprow, 2020-11-06:
nothing to discuss here, closing

closed 2020-11-06 by gkasprow