sinara-hw/FMC_Shuttler #3: Shuttler AFE

dhslichter, 2018-12-05:
Breaking out to a new issue for the AFE design, in response to https://github.com/sinara-hw/Shuttler/issues/2#issuecomment-444438410.

LTC2000 wants a 25 ohm load impedance (40 mA output current for 1 V output).  There is already a 50 ohm termination to ground internally on the chip for each of IOUTP/IOUTN so we want to provide an additional 50 ohm termination to ground for each at the output pins.  Then we route traces over 25-ohm lines (50-ohm differential) to the AFE, using ADA4927.  One option is the solution below, gives 2 Vpk differential output (4V pk-pk) with 10.2 nV/rtHz differential output noise density.  Can't seem to get it much better by playing with components.
[image]

dhslichter, 2018-12-05:
@tprzywoz see above for an initial spec on AFE design, @gkasprow @hartytp check this out too and see if you have disagreements.

hartytp, 2018-12-05:
@dhslichter I'm going to let you lead on this design -- I already have enough on my plate with Sayma and all the EEMs I'm working on and don't have enough head space to contribute in a major way to this design review.

dhslichter, 2018-12-05:
@hartytp ack, I have been doing the same w/r/t Sayma and EEMs....

hartytp, 2018-12-05:
Divide and conquer... :)

gkasprow, 2018-12-05:
There is going to be another AFE outside of the Shuttler board?
We would need some control signals which need to be foreeen.
Did we finally decide which connector we should use? Shuttler can go up to 500MS/s (or even 800MS/s)  per channel, so I'm not really sure if SCSI is good choice here. We can use i.e. QSFP/SAS connectors and cables which have both 8 diff pairs and a few control signals.
With 3 such plugs we will connect all 18 channels + some high speed clock/data lines and a lot of LVTTL signals. The cables are available with various speed grades, so go up to 30GHz (3M Twinax)
You can get them [on cheap](https://www.amazon.com/CABLEDECONN-Internal-36-Pin-SFF-8087-Cable/dp/B00S7KU3PC/ref=sr_1_4?ie=UTF8&qid=1544032408&sr=8-4&keywords=SAS+cable)
External ones are [more expensive](https://www.amazon.com/CableCreation-External-26pin-SFF-8088-Cable/dp/B013G4F5QK/ref=sr_1_8?ie=UTF8&qid=1544032408&sr=8-8&keywords=SAS+cable), but still cheap.

gkasprow, 2018-12-05:
[These](https://cdn.amphenol-icc.com/media/wysiwyg/files/drawing/10117949.pdf) are nice cables. 8 diff pairs + 8 IO wires, rock solid connectors. It is external High Density SAS. Available i.e. [here](https://www.amazon.com/CableCreation-3-3FT-External-SFF-8644-Cable/dp/B01GPCTYEY/ref=sr_1_14?ie=UTF8&qid=1544033318&sr=8-14&keywords=external+SAS). One cay get them up to 10m long

dhslichter, 2018-12-05:
@gkasprow I have no desire whatsoever to try to run Shuttler past ~150 MSPS.  I understand that this doesn't make use of the full bandwidth of the DACs, but I don't want us to end up over-specing other components if there is no intent to go above ~150 MSPS.  I want to just make sure we keep this in mind as we develop.  

That said, I imagine that the crosstalk on SCSI might be higher than we'd want, and that shielded differential pairs might be the best way to go.  I would want to use a reliable, COTS system, and these SAS and HD-SAS cables seem to be one option.  

> There is going to be another AFE outside of the Shuttler board?
We would need some control signals which need to be foreeen.

Yes, there will be some kind of AFE to provide the final gain, filtering, differential-to-single-ended conversion that lives as close as possible to the ion trap.  Control signals might be nice to have if they are for free, but I had at least envisioned a "dumb" front end without things like switchable gain.  I also worry a little bit about digital crosstalk onto the analog signals if we send them down the same cable.  One could break out any digital signals into a separate cable/connector to address this.

gkasprow, 2018-12-05:
These digital signals go outside of twinax cables. They are usually loose wires inside plastic wrap. We can add also external connector, i.e. RJ45 and run LVDS signals over it for possible AFE control.

gkasprow, 2018-12-05:
SAS do not have side signalling, only HD-SAS and mini-SAS have ones.

dhslichter, 2018-12-05:
I would vote for RJ45/LVDS for digital AFE signals, and use the twinax cables for analog only.  Can you confirm the HD-SAS etc have individually shielded pairs?  I can't find a crosstalk spec anywhere (other than "at least 15 dB below insertion loss", which is not very tight), do you know of one?

gkasprow, 2018-12-06:
Yes, they do have individual pairs. They usually use 3M [twinaxial cables](http://multimedia.3m.com/mws/media/673519O/3mtm-twin-axial-cable-sl8800-series.pdf)
I use them in my projects so will measure the crosstalk.

gkasprow, 2018-12-06:
@tprzywoz please measure crasstalk of the twinax cables.

gkasprow, 2018-12-06:
@dhslichter why did you set Rs=25Ohm ? it is current output, so we need 50Ohm load impedance and traces can be 50Ohm as well.

dhslichter, 2018-12-06:
@gkasprow it is current output, but they want us to use 40 mA and have a 25 ohm load for the DAC to keep the output voltages between 0 and 1 V (+/- 1V differential), according to the datasheet.  There is a 50 ohm internal to the DAC, and we could use a 50 ohm transmission line and 50 ohm termination for the amplifier stage.  However, I am a little unclear because the DC path for the current feedback amplifier is such that we would end up running with 576 mV DC offset for the DAC channels, which would end up maybe giving us some issues with distortion as we go to full scale on the DAC.  

If you look in the datasheet, they actually specify the DAC with 12.5 ohm total load (that is, 50 ohm DAC internal, 50 ohm DAC external, plus a 25 ohm load impedance at the front end of whatever following amplifier you're using).  This keeps the voltage swing down and I think ends up helping with the distortion

Here's a version with 25 ohm total load (50 ohm DAC plus 50 ohm at amp), showing what I mean about the DC offsets:
[image]

We can also turn the DAC current down (it can be set in a range between 10 and 60 mA) if that's easier, but it seems that spur performance is best at higher currents.  Anyway, this is something we should test -- I have an eval board but haven't had time to play around with it yet.

dhslichter, 2018-12-06:
regarding twinax, these look good (I remember your showing them before at some point).  It will be great to have a crosstalk measurement for verification but this seems like it would be a good solution.

gkasprow, 2018-12-06:
We have all HW needed to make crosstalk measurement.
I plan to attach 3 SMA connectors to the iPass connector where plug will be inserted.
I want to measure crosstalk between neighbouring pair and pair on opposite side of the connector.
Shall I attach baluns? Or just connect the signal to one of the diff pair wires and observe another?

dhslichter, 2018-12-06:
Ideally, baluns would be used, or single-ended-to-differential amplifiers otherwise.  We are looking for crosstalk at frequencies up to ~100-200 MHz max, so high-speed diff amps would work well also.  

Just for a zeroth-order test, you could try putting a single-ended signal down one wire of one pair and see how much you see single-ended on a single wire of a neighboring pair.  That certainly gives a worse-than-worst-case crosstalk number.