sinara-hw/FMC_Shuttler #4: LVCMOS18 to LVDS - DAC data input

tprzywoz, 2018-12-06:
I made HyperLynx simulation of LTC-2000 data input lines with LVCMOS18 Kintex output. When single resistor was used as a DAC termination all results were nice like here https://github.com/sinara-hw/Shuttler/issues/2#issuecomment-414122354.

And now results with DAC ibis model:
[image]
[image]
Signal on + pin is shifted and also amplitude is reduced by the factor of 2.
I wrote a message to Analog support and i'm waiting for an answer because it could be ibis model problem but also something unexpected in internal termination circuit.

gkasprow, 2018-12-06:
The DAC datasheet says that VCM can be from 0.4V to over 1.5V.
But when we try to set 0.9V VCM it behaves like it was Thevenin terminated to roughly 1.2V inside.

dhslichter, 2018-12-06:
Be interesting to hear from Analog if this is an IBIS bug or if things are truly thevenin terminated inside.  The labels on the graph above are a little unclear -- does "DAC + input" mean pin DBP15?  If you put a resistor between the 0.9V bias and the DBN15 pin, can you figure out how strong the pull to 1.2V is?

tprzywoz, 2019-02-20:
The first answer after 2 months was: "We do not recommend driving the LVDS inputs single ended". I'm 
 still trying to get more information...

gkasprow, 2019-02-20:
I ordered devkit  for LTC2000 so we will check it with Sayma AMC or other FPGA.

dhslichter, 2019-02-21:
thanks @tprzywoz @gkasprow.  Hopefully we can make this work, but if the IBIS model is right then we might be in trouble here.  We could think about using LVCMOS-LVDS bus translators like http://www.ti.com/product/SN65LVDS387 but these require 3.3V LVCMOS inputs (not sure we can get this out of all the FPGA banks) and consume about 280 mW each (!!).

gkasprow, 2019-02-21:
Anyway, even with 1.2V it should work. We will need to pull the matching network slightly up.

gkasprow, 2019-02-21:
@dhslichter the schematics are in the releases. Please have a look.

dhslichter, 2019-02-21:
Will look through schematics and let you know; it will probably be a couple of days before I can do through it all in detail though.

tprzywoz, 2019-02-26:
"The only way to test it to see if it is possible is to build it on the bench. "
So we are waiting for demo board...

tprzywoz, 2019-04-25:
Ok, so some results of test with DEMO 2303A.

Connections:
I connected demo board to Sayma_AMC and added 82R resistor to LTC2000 input (bit 14 - DBN14 and DBP14). It is not best termination but there is no other place to solder:
[image]
Kintex drives hi-z on all DAC data input pins except pin DBN14. DBP14 is connected to hi-z FPGA and also to resistive voltage divider (two 100R connected to DVDD1V8 - generate Vtt 0.9 V ) + capacitor 200nF.  Kintex output at DBN14 is LVCMOS18, DRIVE = 8, SLEW => SLOW.

Results:
[image]
[image]

I could change output voltage so we can drive LTC2000 with LVCMOS. There is a problem with oscillations when FPGA switches output but in my opinion it is only a consequence of the implemented "termination" (remember that distance between 82R termination and DAC is about 3 cm and I use resistive divider - not stable Vtt=0.9V).

gkasprow, 2019-04-25:
take into account that DAC is driven with 150MHz rate. So with such a poor matching, this looks good.

dhslichter, 2019-04-26:
Seems like a reasonable proof of principle!  What is the impact on power dissipation from the VT=0.9V on DBP14 drawing current from the DAC inputs?

tprzywoz, 2019-05-20:
Do you mean what is the impact of DAC inputs on VT circuit? I made another test: both FPGA outputs were hi-z, then I connected Vt to DBP14 with an ammeter. The result was 0 A.

gkasprow, 2019-12-22:
@tprzywoz please publish recent results that you included in your master thesis.