sinara-hw/FMC_Shuttler #6: Shuttler schematic review

tprzywoz, 2019-02-20:
I've just published schematics of Shuttler. I know that Sayma review must be done first to start Shuttler review, but I will be grateful for checking compliance with the expected specification -  I want to finish my master's thesis as soon as possible :)

gkasprow, 2019-07-24:
@tprzywoz Is this the last version of schematics? I want to introduce some changes and also do the review.

tprzywoz, 2019-07-24:
No, I changed the organization of FPGA schematics and fixed Sayma 2.0 issues that also affected Shuttler. I still have to finish some details. I'm going to push new version tomorrow.

tprzywoz, 2019-07-25:
I published the last version. What changes do we want to make? 
My questions about potential changes:
As I understand a direct PORT0 connection to GTH is already working.  Can we remove ETH_PHY? / Do we want to wait for Sayma 2.0 to be sure? 
Same question about Exar module.

gkasprow, 2019-07-26:
Let's leave as it is now. Do we miss some GPIOs?

tprzywoz, 2019-07-26:
If we want to implement EXT_CLK_IN / WR_REF_CLK_IN connected to FPGA then we need more IOs.
Also I thought about digital signals to control external AFE. Currently, we have 10 x 1.5V single-ended lines. We could add 3.3V lines without PHY.

gkasprow, 2020-08-14:
not relevant since we moved to FMC

closed 2020-08-14 by gkasprow