sinara-hw/FMC_Shuttler #7: missing decoupling capas at the IC72

gkasprow, 2019-02-21:
There are FB connected to the supply pins. Aren't there missing decouplig caps?

gkasprow, 2019-02-21:
can the VOCM be left dangling?

dhslichter, 2019-02-21:
VOCM for the ADA4927s needs to be tied to ground.

tprzywoz, 2019-02-21:
> VOCM for the ADA4927s needs to be tied to ground.

Done

> There are FB connected to the supply pins. Aren't there missing decouplig caps?

Added according to datasheet: 1nF 100nF ceramic and 10uF tantalum capacitors for each supply.

dhslichter, 2019-02-21:
> Added according to datasheet: 1nF 100nF ceramic and 10uF tantalum capacitors for each supply.

Please make notations for voltage and dielectric spec (X7R or NP0) for the ceramic caps.  These amps may be driving a reasonable amount of current into the terminated transmission lines and so we want to make sure we don't have dc bias issues with the capacitors.

gkasprow, 2020-11-06:
it's not relevant anymore. we don't have any caps in the signal path in the FMC project.
I added a note on the AFE schematics

closed 2020-11-06 by gkasprow