sinara-hw/FMC_Shuttler #10: DAC selection

gkasprow, 2019-12-22:
@tprzywoz finished layout of the board and want to close his master thesis. We decided that the main topic would be the optimization of the DAC connectivity that optimizes the price. He cannot wait longer for physical prototypes.
So, the question is what DACs should we choose? Do we stay with LTC2000 or can switch to a lower-cost alternative. I'd like to freeze the specification in early January.

dhslichter, 2019-12-23:
@gkasprow you should proceed however you see fit; we will still have to discuss the specification and I don't know whether this will be complete and ready to freeze by early January.  I think it is highly likely that we will NOT use the LTC2000 because of cost/power/complexity.  I am currently planning to target AD9117 as the DAC choice, with digital core at 1.8 V and analog at 3.3 V, tuned for 20 mA output current.  However, all of this is subject to discussion/initial testing/etc.  If you want to choose this DAC and freeze that spec for Shuttler 0.1, so that @tprzywoz can complete his thesis work, that is fine with me, however it is likely that we will have to come back and make various adjustments to do a proper Shuttler 1.0 design.

gkasprow, 2019-12-23:
We can switch to AD9117, I don't want to produce something that will be useless later on. What else would you like to change?

dhslichter, 2019-12-24:
Let me put up a draft spec in a day or two once there has been time for comment on my email to you and M-Labs.

gkasprow, 2020-11-06:
My colleague implemented dithering on this DAC devkit, he will test it and characterize within a 2..3 weeks. We can close it.

dhslichter, 2020-11-07:
@gkasprow we have been able to achieve 18 monotonic bits using dithering on a devkit for the AD9117 in tests at NIST before the pandemic hit.  The temperature coefficient is less well characterized, but appears to be helped by using an external reference.

gkasprow, 2020-11-07:
can you write more about the results and methods?

dhslichter, 2020-11-13:
Went into the lab and took some measurements today to verify the previous work.  The plot below shows the output voltage measured by a high-precision DVM with a slow time constant (~1 second) to reduce the noise.  There is a lot of pickup in the lab and the setup is definitely not optimized for noise, but this lets us see the trend at least.

The values here represent dithering around a major carry (digital codes 4095-4096) as well as with just the LSB flipping (codes 4096-4097).  The waveforms are very simple.  We output a looping digital waveform where for each 16 samples, the last (2,4,6,8,10,12,14) samples have a code one larger than the rest (e.g. 12 samples of 4095 followed by 4 samples of 4096 would be an average of 4095.25).  For some of the waveforms it was done in chunks of 32 samples rather than 16 samples, doesn't seem to make a difference with this extremely slow averaging.  

In the end, this lets us get 3 extra effective bits that are nearly monotonic.  My student took some data several months ago showing 4 extra monotonic effective bits, but these were much closer to 0 V, and so less affected by gain errors/drifts.  These data are at the midpoint between 0 V and full scale (+/-1 V).  

[image]

dhslichter, 2020-11-13:
Also took some measurements of temp coefficient.  I set up the DAC eval board with a temperature sensor IC bonded on top of the DAC chip with thermal paste, and then gently heat gunned the underside of the board and the surrounding box to get the temperature up to ~45-50 C.  Then I stopped the heat gunning and just recorded the output voltage along with the temperature of the sensor as the board slowly cooled back down to room temperature.  For an output of +1 V or - 1 V (full scale), I measured a voltage drift of 20 uV/C, with opposite signs (thus a gain drift of 20 ppm/C).  For an output of 0 V, the voltage drive was 1 uV/C.  These numbers are consistent with the datasheet, which claims a max gain drift of 40 ppm/C and a much smaller offset drift.

dhslichter, 2020-11-13:
The fluctuations of the output voltage at full scale on ~second timescales are fairly large, 10s of ppm, but this setup has basically not been optimized for noise at all.  I am a little nervous about these fluctuations, but I see so much pickup and switch mode spikes and such when looking at the output on an ac-coupled scope that I don't know how worried I should actually be.  It also appears that these fluctuations are smaller after some warmup period of 15 minutes or so.  This is something that will require much more careful lab setup and measurement if we want to get to the bottom of it.

gkasprow, 2021-01-24:
nothing to do here, closing

closed 2021-01-24 by gkasprow