sinara-hw/FMC_Shuttler #12: Shuttler FMC

kaolpr, 2020-06-04:
During the conversation on HDL/SW specification, an idea of having Shuttler moved to the FMC modules has arisen. Let the authors speak:

> @dhslichter FWIW, although I have been occupied with other things these past couple of months, my current plan is to try to make a Shuttler design that is EEM/Kasli compatible, similar to how Phaser has been developed. If others are interested in pursuing the specifications above, perhaps we should branch or rename the EEM version to something else?

> @gkasprow Several questions arrive:
> * would you need DRTIO? In new CPCIS form-factor, together with ZUS+ the DRTIO in CPCIS will be supported. However, in EEM we would have a faster version of Fastino. Of course, we can connect DRTIO over SFP
> * How many channels? We won't fit 24 probably. On the other hand if we us
> * XCKU035-1FFVA1156C, it's the same chip as CERN is using here. We will get very good pricing, comparable with Artix
> * what is the use case? how many channels would you need in total?
> * if we go for CPCIS natively, we would get extra real estate and 24 channels would be feasible.

> @dnadlinger Going for EEM would mean reduced channel count; I am also looking at ways to reduce power consumption by choosing different DACs and amps. We were doing some DAC testing in the lab before the quarantine but that has been stopped, unfortunately.
>
> If the CPCIS form factor works, that's great, but I am hesitant to be the guinea pig here. My preference would be to develop a board as an EEM, which would allow us to get up and running, and ease the debugging process (as with Fastino and Phaser). Down the road, if the design is proven out and there are advantages to using a single card as a DRTIO satellite, rather than running over EEM from Kasli, one could port a lot of the design over.
>
> Total channel count desired would be something like ~100 for a given setup, possibly a bit more. I think the primary limitation will be power dissipation in the rack, and thermal management. I don't think we need 24 channels per card, 12-16 would probably be just fine.

> @gkasprow For the development of the Shuttler HW & GW we received already the research grant. It is clearly stated that it should support MTCA so we must somehow cope with it.
What if we make it in FMC form-factor? Then we could use it with low cost [AFC](https://ohwr.org/project/afc/wikis/home) carrier and also with [CPCIS FMC carrier](https://ohwr.org/project/diot/wikis/diot-fmc-carrier). AFC/AFCK is well-tested solution (a few hundreds in operation), and now we are working on 4-th revision together with a few research institutions (CERN, LNLS, GSI, WUT).
The CERN CPCIS carrier uses very low cost (special Xilinx discount) Kintex US FPGA, already supported by ARTIQ.
With such an approach we could further fund the Shuttler development, meeting both funding agency goals as well as your requirements. We are working intensively with CERN on CPCIS, have suitable funding so soon the entire EEM ecosystem will be migrated to this ecosystem keeping full compatibility with existing EEM modules.
I already proved that [16-channel * (ADC + CFD + TDC)](https://github.com/elhep/FMC_ADC100M_10B_TDC_16cha/wiki) together with SAS connector is feasible on single FMC HPC.
I think 12-16 DAC channels per FMC can be done easily. I did a simple trick - the power section is placed on tiny mezzanine plugged to the FMC board and secured with 2 screws. Both boards are produced simultaneously with no additional cost.
With such approach, we will lower the cost significantly due to the economy of scale of AFC. The design and verification of such DAC would also be much lower risk and cost.
What we can do is to design two FMCs - one with LTC2000 and another with lower-grade DAC and see what approach is feasible in a real application.
In the past I already did 16-channel [14-bit FMC DAC](https://ohwr.org/project/fmc-dac100m14b16cha-adc2m14b4cha/wikis/home) and even managed to fit 4 ADC channels and some IOs.

kaolpr, 2020-06-04:
Ops. Sorry for that.

dhslichter, 2020-06-12:
Doing developments on FMC seems to be a sensible idea.  One can then have the choice of different "carrier" boards as appropriate for MTCA/EEM/CPCIS, or just use a standard FPGA eval board to get things up and running (e.g. KC705).  I think a lot more of the questions come around what one chooses for the DACs, amps, etc.  Thinking about power budget, space constraints, and so forth.

closed 2020-11-06 by gkasprow