sinara-hw/FMC_Shuttler #13: Shuttler DAC/amp design

dhslichter, 2020-06-12:
Proposal for Shuttler DAC and amplifier selection/design @gkasprow @kaolpr :

- DAC: [AD9117](https://www.analog.com/en/products/ad9117.html).  For output tones in the DC-15 MHz range (target), has very good noise/distortion/spurs.
    - two channels
    - 14 bits
    - 125 MSPS
    - 85 dBc SFDR, 82 dBc IMD for 10 MHz output tone
    - ~0.5 LSB DNL/INL spec after self-calibration (DNL tested OK, INL questionable)
    - -163 dBm/Hz noise floor @ 1 MHz, -157 dBm/Hz @ 10 MHz
    - use external reference, 1.25 VDC, ultra low noise and tempco desired.  Suggest [LTC6655](https://www.analog.com/en/products/ltc6655.html) as used in Fastino.  Gain tempco of DAC is spec'ed on data sheet at 40 ppm/C but this is with internal reference, which is not very good.  Measurement of gain tempco with good external reference still needed.  
    - Use 3.3 V AVDD, 1.8V DVDD for reduced power consumption/improved SI (digital I/O is LVCMOS).  Using LVCMOS18 for digital interface also allows use with US FPGA or 7 series FPGA (no HR banks required).
    - 14-bit LVCMOS parallel data input bus runs DDR.  Some care is required to ensure deterministic latency from power-up to power-up, given de-interleaving re-timer (see datasheet).  
    - SPI access also required for register programming
    - Estimated power dissipation ~100 mW/channel

- output: differential-to-differential output amplification with gain~5, made from one [ADA4896-2](https://www.analog.com/en/products/ada4896-2.html) per channel for good matching and very low noise.  
    - 1.36 nV/rtHz, input referenced, including resistor noise effects, at G = 5 (see datasheet Table 13 for recommended resistor values)
    - amplifier GBW gives natural rolloff, might include pads for additional capacitor in feedback network to give further suppression of Nyquist image feedthrough if desired (option)
    - choose supply voltages to minimize dissipation (3 mA quiescent current per amp, 6 mA/DAC channel)
    - capacitive loads cause ringing, need some small series resistance to avoid problematic interactions with output cable capacitance

- board output would be differential voltages from output of amp stage, to be carried on 100 ohm differential pairs to remote AFE board.
- differential-to-single-ended conversion, additional gain, additional filtering, line termination, etc all on remote AFE board

- needs discussion/measurement:
    - DAC output/amplifier circuit topology and component values 
    - DAC tempco measurements with external reference
    - output cable/connector specification
    - line termination scheme for remote AFE (will ac termination work OK?  reduces static dissipation, but introduces phase shifts)
    - choice of rails for output amplifiers

gkasprow, 2020-06-13:
For connectivity purposes, I'd like to use mini-SAS (internal ones since we have limited area on FMC). They have a 4-wire side channel that normally run SPI (HDD blinking LEDs). Cables are cheap and individually shielded, specified up to 12Gb/s. I use such cables in various analog designs. 
What about supplying the remote AFE?
I did initial schematic and PCB placement, it looks feasible
[image]
[image]
The right-most PCB is a mezzanine with all supplies. I used this approach already and works fine. I will reuse already working module which delivers several low noise rails 1.8VA, 3.3V, 3.3VA,  5VA, -3VA
https://github.com/elhep/FMC_ADC100M_10B_TDC_16cha
[image]

gkasprow, 2020-06-13:
such FMC supports 16channels. Two double DAC chips are connected to same bus via level translator-buffer. It enables compatibility with both 2V5 and 3V3 VADJ boards.
The DACs have separate data and sampling clocks and require low Ts/Th of 0.25ns/ 1.2ns, so we have time to latch 4 data cycles per 8ns clock cycle. I want to use 74AVC20T245. Such buffers usually have skew below 1ns. In case of issues, we can run the board at 100MHz. This will probably require some calibration of clock phase using DCM or delay.
Another thing is the jitter. I assume we will operate it in wide-band mode, far away from sampling frequency, and use dithering, so clock jitter generated by FPGA is not an issue.

gkasprow, 2020-06-13:
It looks like we can also use DDR mode which will simplify timings a lot and use full DAC speed.
We have 34 * LVDS for LA
24 * LVDS for HA
22* LVDS for HB
Which gives 160 single lines.
To connect 16 DACs, we need 8 * 16 = 128 lines
Even if we restrict them to bank/clocking zones, we still manage to fit 16 channels.
Some XIlinx devkits have missing some HA and HB lines, but KC705 has them all; the AFC board can also work stand-alone.

gkasprow, 2020-06-14:
One word about the level translators. We actually don't need them because DACs work with 1.8...3.3V range. But I want to use them to isolate DAC from FPGA.

dhslichter, 2020-06-14:
Looking good, @gkasprow!  Some comments on the above:
- 16 DAC channels on an FMC would be high enough density for my purposes
- Let's plan to use DDR mode @ 125 MHz on the DACs, I would like to be able to run them up to full speed (125 MSPS/channel).  
- Those level translators (SN74AVC20T245) don't look like they will run fast enough if translating to 1.8 V?  Datasheet quotes 210 Mbps at 1.8V output.  I prefer the idea of running the DVDD for the DACs at 1.8V, with the AVDD at 3.3 V.  We want the higher AVDD rail for better performance of the analog outputs, but can save ~25 mW per chip by using the lower DVDD rail (200 mW total for the board).  I also like the reduced potential for digital-to-analog crosstalk if we are using 1.8V DVDD.  
- Why do you want to use a level translator?  I don't understand the need to isolate the DACs from the FPGA, sorry.  It seems to me that the potential for skew/timing problems is something we should avoid if we can.  Also, we will want to have a proper clock distribution system for the DAC clock, not using these translators, to get the best performance.  
- Requiring a carrier board that has a fully populated HPC FMC (in order to run all the DACs) is fine with me.  KC705 has a fully populated HPC FMC so would work for this.  
- I like the idea of a power mezzanine, but suggest we need some adjustments compared to the version you linked to above.  The rails we need are 3.3VA, -3VA, 1.8VD.  We want the lowest noise we can get here, as the experiment will be very sensitive to DAC noise.  Suggest LT3094 for -3V rail (to follow filtered buck reg TPS63710, as on current power mezzanine), and 2x LT3045 (one for DACs, one for amps) to supply +3.3V rail.  
- suggest considering shielding cans for DACs?
- mini-SAS cables sound fine, side channel would be good to wire up for SPI communications (4 wire) between FPGA and remote AFE.
- remote AFE will have its own power supply.  Likely we will also break the grounds at the remote AFE (or at a minimum, maintain that option); since signals are fully differential coming out of the FMC, we can use a suitable differential-to-single-ended amp on the remote AFE to allow the grounds to float relative to each other within some acceptable range.

dhslichter, 2020-06-14:
Also: because DAC outputs are centered around 500 mV, we need a very low-noise 500 mV reference for the op amp noninverting inputs, so that the swing of the op amp outputs is correct for the gain.  It's OK if the output has a nonzero common mode voltage at DC, since the remote AFE will strip this off anyway.

gkasprow, 2020-06-14:
>    16 DAC channels on an FMC would be high enough density for my purposes
good

>    Let's plan to use DDR mode @ 125 MHz on the DACs, I would like to be able to run them up to full speed (125 MSPS/channel).

OK

> Those level translators (SN74AVC20T245) don't look like they will run fast enough if translating to 1.8 V? Datasheet quotes 210 Mbps at 1.8V output. I prefer the idea of running the DVDD for the [...]

I had issues with 16bit ADC coupled with FPGA. The FPGA generates a lot of ground-bouncing noise. It adds up to the digital lines. The result was that I lost 2 LSBs and got funny patterns on the image (the ADC was part of CCD camera electronics). We can try without buffers in case of these DACs, because they have 14bits and don't have a fragile sampling and hold circuit.

That's why I was asking about jitter. If all DACs will run with the same frequency, I will add proper clocking, which will also deliver copy of clock to every FPGA bank used for DAC control.

   > Requiring a carrier board that has a fully populated HPC FMC (in order to run all the DACs) is fine with me. KC705 has a fully populated HPC FMC so it would work for this.

> I like the idea of a power mezzanine, but suggest we need some adjustments compared to the version you linked to above. The rails we need are 3.3VA, -3VA, 1.8VD. We want the lowest noise we can get [...]

we will produce the supply mezzanine anyway, together with FMC. The existing design uses ADM7171 and ADP1740 LDOs for positive rails
[image]
And such circuit for negative rail. 
[image]

The system was 8-bit so noise was not that important.

>    suggest considering shielding cans for DACs?

Do you mean to shield it against power supply mezzanine noise? We can make the bottom layer as a solid copper. I'd rather add it to the AFE circuit, but don't think it's necessary.
I can try if there is a place for it. If we scrap the digital buffers, the shield should fit.

>    mini-SAS cables sound fine, side channel would be good to wire up for SPI communications (4 wire) between FPGA and remote AFE.

We would have 4 wires per each 8 channels

 >   remote AFE will have its own power supply. Likely we will also break the grounds at the remote AFE (or at a minimum, maintain that option); since signals are fully differential coming out of the FMC, we can use a suitable differential-to-single-ended amp on the remote AFE to allow the grounds to float relative to each other within some acceptable range.

If we break the GND, additional passive CMC may be necessary

>Also: because DAC outputs are centered around 500 mV, we need a very low-noise 500 mV reference for the op amp noninverting inputs to make a proper fully differential amplifier circuit. It's OK if the output has a nonzero common mode voltage at DC, since the remote AFE will strip this off anyway.

So, we could avoid -3V rail ?

dhslichter, 2020-06-15:
> we will produce the supply mezzanine anyway, together with FMC. The existing design uses ADM7171 and ADP1740 LDOs for positive rails

I saw that schematic already, and the noise needs to be lower.  That's why I requested those other LDOs instead.  I would suggest just making a variant of that mezzanine (same board size/footprint, but different part numbers) that will meet the needs of this board better.

> So, we could avoid -3V rail ?

No, I don't think so, because we want a gain of ~5 on Shuttler itself.  This means that each of the differential outputs (after amplification) swings +/- 2.5 V from the output common mode voltage.  We want something like this amount of gain on Shuttler (in other words, not putting it all on the remote AFE board) to help mitigate noise pickup which may occur between Shuttler and the remote AFE, and because this level of gain represents a nice compromise in terms of graceful bandwidth rolloff (and avoiding peaking), input-referred noise, and power dissipation (due to the supply voltages needed).  It also reduces the amount of gain required on the remote AFE, which could then be a single-stage amp instead of needing two stages.  

> If we break the GND, additional passive CMC may be necessary

Why?  I am not opposed necessarily, but I feel vaguely like they have caused some difficulties in other projects, right?  

> We would have 4 wires per each 8 channels

I think that's plenty.  Depending on what one wants to do with the remote AFE board, these could be used to configure or read back data from a number of different things (analog switches, controllable gain, calibration ADCs, etc).  These are the HD mini-SAS, correct?

> Do you mean to shield it against power supply mezzanine noise? 

The shielding would be for power supply noise from the mezzanine, or digital switching noise from the nearby FPGA, or from neighboring cards in the rack, or....basically, this is my paranoia wanting to be careful with these signals.  Careful routing of the analog signals should probably already win us a lot (internal layers with ground planes top and bottom and vias on the sides, for example).  

> If all DACs will run with the same frequency, I will add proper clocking, which will also deliver copy of clock to every FPGA bank used for DAC control.

Yes, all the DACs would run at the same frequency.  This sounds good.

dhslichter, 2020-08-23:
Why do we have a P5V0 rail?  This will be the positive supply rail for the output op amps, right?  That can be 3.3 V for reduced power dissipation.  The DACs have a V_CM of 500 mV and swing between 0 and 1 V; with G=5 the output signal will have a common mode voltage of 500 mV again, with output voltages swinging between +3V and -2V.  The op amps are rated to run with 200 mV headroom on the rails, so this would still be OK.  We could move the 3.3 to 3.5 V, for example, if we were worried.

gkasprow, 2020-08-24:
I want to have a separate rail for opamps. It does not have to be 5V. We will reduce it to something like 3.5V.
The opamps have usually poor performance close to the supply rails. If we finally decide and test that 3.3 is enough, we can remove this rail.

dhslichter, 2020-08-24:
I think we should have a separate rail for the op amps -- I was just suggesting that we reduce the voltage from 5 V to ~3.5V.  We can have rails at -2.5 V and 3.5 V so that we have 500 mV headroom either direction, and can increase both directions a little if we need to.

Another reason we need a separate rail is that a single LT3045 won't be able to run all the DACs and all the op amps -- it only does 500 mA max.  The op amps will draw 16 * (2*3) ~ 100 mA quiescent current, and will be drawing more based on the terminations at the AFE.  The DACs draw (55 * 8) = 440 mA on the AVDD rail.  One rail for DACs, one rail for op-amps sounds good.

gkasprow, 2020-11-15:
@dhslichter did U have a chance to check the FMC card design? I'd like to start routing it ASAP because the project deadline is close.

dhslichter, 2020-11-17:
@gkasprow sorry slammed but will do tonight (USA time)