sinara-hw/FMC_Shuttler #14: AFE supply

gkasprow, 2020-06-13:
What are the plans for supplying the  remote AFE?

dhslichter, 2020-06-15:
Remote AFE is a separate board entirely, with its own independent power supplies.  We just need to specify the interface between the remote AFE and the Shuttler FMC card for the analog signals, as well as for the digital (SPI) signals in the side channels of the cables.  

My proposal:
- analog signals are provided over external mini-SAS
- analog signals are differential, with nominal common-mode voltage 500 mV (referenced to FMC ground)
- remote AFE should be configured as instrumentation-amplifier-type circuit to perform differential-to-single-ended conversion, provide gain, and tolerate common-mode offset of up to ~few V (high-impedance inputs, allows potential use of separate ground for remote AFE/ion trap that is not closely coupled to computer/FPGA ground)
- remote AFE provides differential termination (100 ohm, possibly ac coupled with knee at ~1 MHz).  
- digital lines in side channels are all opto-isolated at remote AFE with DIP-switch-selectable line directions.

gkasprow, 2020-06-16:
sounds good. What supply range do you need?
Would you like to supply it from [Aux Power Suply](https://github.com/sinara-hw/meta/issues/48)?
Or from standard 12V wall wart (bad idea); or, some low noise +/- 12V bench supply?

dhslichter, 2020-06-17:
Let's defer the discussion of the remote AFE for the immediate time being.  My vision is that this should be a fairly simple "physicist"-type board.  We will likely want to put some reasonable consideration into the power supply, including noise testing.

gkasprow, 2020-06-17:
What voltage ranges do you expect at the ion trap electrodes?

dhslichter, 2020-06-19:
It will depend on the trap!  That's why we should leave the remote AFE as a separate task and not worry too much for now.  I'd say a "generic" value people have used in the past is +/- 10 V, but some may want +/- 20 V, +/- 50 V, ....

dhslichter, 2020-06-19:
The point is that different traps may call for different AFE designs.  I would like to leave the remote AFE as completely separate now and not worry about it, other than specifying 1. the connector it must use, 2. the termination for the analog signal, and 3. the ground potential difference it can tolerate relative to the Shuttler card.

gkasprow, 2020-06-19:
True, but we need something at the end of the cable to properly characterize the FMC design. Something simple that meets at least a typical use case.

dhslichter, 2020-06-19:
For a simple output board, use something with +/- 12 V or +/- 15 V output rails (could be moved up or down a bit as needed), with differential-to-single-ended conversion and a target gain of 2 or 2.5.  This would give a total output swing between +/- 10 V (or +/- 12.5 V).  A typical instrumentation amplifier or difference amplifier probably won't work because the bandwidth is not high enough (we would like ~15+ MHz bandwidth ideally).  Probably will need to build the circuit out of discrete op amps such as [LM6172](https://www.ti.com/product/LM6172) or [THS3121](https://www.ti.com/product/THS3121).  These have sufficiently high slew rate, low noise (including at low frequency -- we care about the 1/f voltage and current noise below 1 kHz), and low quiescent current.  Configure as a 3-op-amp instrumentation amplifier?

dhslichter, 2020-06-19:
The [LT1995](https://www.analog.com/en/products/lt1995.html) is another option with pin-strapped gain and internal resistors.  The resistors are not super well-matched, and the quiescent current of 9 mA means ~300 mW quiescent dissipation PER CHANNEL at +/- 15V supplies.  I would prefer to use discrete op amps with matched resistor arrays like this https://www.digikey.com/product-detail/en/vishay-beyschlag/ACASA1001S1001P100/749-1015-2-ND/4725768 which have 0.05% resistance match and 15 ppm/C tempco match for less than $1/channel, to build the difference amplifier stage.  Can use one resistor array for the buffers w/ gain, another for the subtractor stage.   Using +/- 15 V supplies with +/- 10V output voltage gives us some headroom to allow for common-mode voltage difference between the remote AFE and the FMC card.  Using a CMC on the input of the remote AFE would help raise the CMRR further at high frequency.

gkasprow, 2020-06-20:
Athat about the AFE form factor? Some Hammond extruded enclosure? What output connectors to the trap?

dhslichter, 2020-06-20:
Output to the trap: 25-pin d-sub for now, people can think about if they want to get fancier later.  
AFE form factor: yes, probably would live in some cast aluminum box (not extruded).

dhslichter, 2020-08-23:
> Let's try to fit into the smallest one
> 100x50mm

Heat load will be a major consideration here.  Each LM6172, with +/- 15V supplies, has 138 mW quiescent dissipation, and we will use 1.5 per channel.  With 16 channels, this is 3.3 W quiescent dissipation.  This will of course be increased if there is lots of slewing going on.  I think it would be good to plan for some mechanism of heat sinking the op amps directly to the case.  We will also want to use a matched resistor array with a very good tempco matching between resistors, so we maintain good common-mode rejection over temperature.

gkasprow, 2020-08-24:
what about attaching the box to the optical table? Or, you don't want to dissipate heat there to keep the optics temperature stable. We can also attach the heat sink to the enclosure bottom and make good contact between PCB and the box wall. 
In such case we need a bigger box since components will be mounted on one side

dhslichter, 2020-08-25:
In general you don't want to use the optical table as a heat sink.  My point was that yes, we may need a bigger box because we don't want the AFE to get too hot and that means heat sinking to the box itself.  It might be preferable to have a custom-machined aluminum pedestal for the PCB, (this can be done very cheaply) that then bolts to the inside of the box.  One could have large exposed (no silkscreen) metal ground pads on the backside that are then held in contact with the pedestal using screws.  You can carve out cavities in the pedestal for components on the back side, through-hole pins, etc.

gkasprow, 2020-10-20:
LEt's make it compatible with the [AUX_PSU](https://github.com/sinara-hw/AUX_PSU)

gkasprow, 2020-10-20:
@dhslichter This is the first iteration of possible AFE. We can use standard Hammond 1590BB2F die cast aluminium box.
[image]

The PCB is bolted to the laser-cut 2mm metal sheet that is later on bolted to the die-cast box. Only THT and some RC holes are cut in the sheet. This is super-cheap, less than 1$ per such sheet.
The upper cover has cutouts for connectors. It's also cheap to machine them in such way.
The LEDs are only for testing - the DIOs are optically isolated and drive the SPI register which controls LEDs.

gkasprow, 2020-10-20:
what about such AFE schematic?
[image]

gkasprow, 2020-10-20:
I assume you don't need variable gain or switchable filter.

dhslichter, 2020-10-22:
- At least for now, we don't need variable gain or a switchable filter.  This can always come later on some fancier designs if people want it.  I think resoldering is probably the best option rather than trying to implement hot switching, but it's worth thinking about in the longer run.
- I would use an ac termination for the input, otherwise we will have a needlessly large current draw on the Shuttler card itself.  I think at a minimum we should have pads to allow either RC termination or R=100 ohm termination (can just put a 0R on the C pads).  For starters, we might set the corner frequency at ~2 MHz (should be OK, assuming few-meter cable lengths), and one can always adjust this later in the testing stages by changing the capacitor value.  
- It seems that R3 and R5 should match R1 and R8, and R6 should be larger?  Currently it looks like the gain of this circuit is ~100, we want gain ~2 or 2.5 IIRC (depends on exact gain on FMC card)

- One thing we could considering placing would be the ADC we are contemplating using to allow for calibration/slow feedback on voltages.  I would recommend using something like the [AD4115](https://www.analog.com/en/products/ad4115.html), which could basically be used to monitor all 16 channels and send measured values back over SPI.  
    - AD4115 can handle voltages in the range +/- 20 V on each pin (single-ended measurements).  No external components required.  Could read directly from the output of the buffer amplifier (before R4/C5, to eliminate errors due to voltage division with ~1 MOhm input impedance of ADC).  
    -  In order to correct for both gain and offset in the Shuttler DACs (e.g. due to temperature drifts), one would ideally be able to make measurements with several different output voltages.  However, if you have ions in the trap this might not be possible (i.e. voltages have to stay the same, or within some small range).  Therefore I think it might be useful to have an analog switch to disconnect the output connector from the AFE circuit.  The filter capacitors on the trap will maintain their voltages reasonably well for the duration of each calibration (~some milliseconds).  A relay would be best for this (low on resistance, high off resistance, no charge injection) but may not fit.  Analog switches are another option but on resistance and charge injection are important considerations.  The AD4115 has 4 programmable digital output pins which could be used with a 4:16 decoder circuit to control which switches are opened to allow DAC calibration.  All programmable over SPI, makes things simpler.  

    -  The AD4115 has internal registers to allow compensation for ADC offset and gain error (one-time calibration).  It might be useful to have test points (doesn't need to be panel-accessible) to connect a laboratory precision calibrated voltage source for each of the ADC channels, so that one could calibrate the offset and gain for each ADC channel and send this information along on a datasheet for the board.  Tempco for gain and offset are very small (few ppm) so these calibrations would probably be

dhslichter, 2020-10-22:
output connector - is this 25 pin D-sub?

dhslichter, 2020-10-22:
In terms of relays to enable the DAC calibration sweep, I would recommend something like this part: [Pickering 109-1-B-5/2D](https://www.pickeringrelay.com/product/109-1-b-5-2d/).  SPST NC reed relay, switch/bounce times <1 ms, very small form factor (can fit 16 channels into 64 mm x 15 mm board area, we would just need to size up the Hammond box slightly), operates from 5 V with 7 mA, so we can drive directly from TTL/CMOS decoder, integrated diode.  A little pricey at $7 each, but I think we will want the ability to do frequent automatic calibrations on Shuttler.  One can make a board variant where the relays are bypassed by a jumper, so people can DNP and save money if they want.  We can get less expensive relays (<$2 each, saves $80/board) if we are able to use a bit more board space (100 x 20 mm) and use 1 Form A relays (normally open), which would mean dissipating 800 mW for keeping all the relays closed all the time unless calibrating.

gkasprow, 2020-10-22:
The connector is a high-density 26pin DSUB. Any preference here?

gkasprow, 2020-10-22:
@dhslichter there are also [OPTO-MOS](https://datasheet.octopart.com/AQY212EHAX-Panasonic-EW-datasheet-13536469.pdf) switches that have very low charge injection due to lack of steering circuit connection. It's a LED + solar cell + FET. 1.2$/pc

gkasprow, 2020-10-22:
Do you mean having one output connector for external precise DMM and 16 relays to selectively connect it to the AFE outputs?
And yet another 16 relays to disconnect the AFE from the trap?

gkasprow, 2020-10-22:
These [relays](https://www.pickeringrelay.com/product/116-1-a-5-2d/) are smaller and cheaper

dhslichter, 2020-10-22:
> Do you mean having one output connector for external precise DMM and 16 relays to selectively connect it to the AFE outputs?
And yet another 16 relays to disconnect the AFE from the trap?

No, I had in mind 16 relays to disconnect the AFE from the trap, and 16 test points (between output buffer and relay, on each channel), for a precision DMM (or a precision input source, alternatively, with output buffer powered off but ADC powered on).  This would only be accessible with the box open, you just do the calibration once as part of board test.  Another way to do this would be to have an external testing jig that plugs into the output connector, with switchable relays on that testing jig to connect to a DMM.  The issue there is that one has a voltage divider effect with the series output resistance of the RC filter and the DMM input impedance, which could be noticeable at the 16-bit level depending on DMM input impedance (10 MOhm vs 47 ohm = 4 LSB, for example).  If you have a >1 GOhm input impedance DMM then this is the simplest way to go, though, and you don't need the test points.  

> @dhslichter there are also OPTO-MOS switches that have very low charge injection due to lack of steering circuit connection. It's a LED + solar cell + FET. 1.2$/pc

If an OPTO-MOS switch has low enough charge injection at the level we care about (we need charge injection to be ~10s of pC maximum, ideally less, over the entire range of source and drain voltages of the switch), it could work.  We also want the resistance to be low (<1 ohm is fine) e.g. AQY211EH.  We also want the on-state resistance to be independent of drain and source voltages; I suppose you get this with an opto-isolated relay but not with a typical semiconductor analog switch.  

> The connector is a high-density 26pin DSUB. Any preference here?

Mostly it comes down to whether there are good, well-constructed shielded cable assemblies available.  Have you had good luck finding these with this type of connector?

> These relays are smaller and cheaper. And fit nicely

Yes, those were some of the ones I was looking at.  We will dissipate 800 mW steady state just running them, but if the rest of the board is already dissipating 3.3 W steady state maybe this isn't the end of the world.  These are slightly smaller footprint than the OPTO-MOS but it's similar.  

Board design with relays looks nice, seems like things fit well!

dhslichter, 2020-10-22:
Another consideration with the optical relay is what the capacitance to ground is -- appears that there is a substantial voltage dependence of capacitance to ground, which would cause signal distortion.  If so we should avoid this.

gkasprow, 2020-10-22:
> > Do you mean having one output connector for external precise DMM and 16 relays to selectively connect it to the AFE outputs?
>> And yet another 16 relays to disconnect the AFE from the trap?

> No, I had in mind 16 relays to disconnect the AFE from the trap, and 16 test points (between output buffer and relay, on each channel), for a precision DMM (or a precision input source, alternatively, with output buffer powered off but ADC powered on).  This would only be accessible with the box open, you just do the calibration once as part of board test.  But now I think I see this is dumb, and a better way to do this would be to have an external testing jig that plugs into the output connector, with switchable relays on that testing jig to connect to a DMM.  The issue there is that one has a voltage divider effect with the series output resistance of the RC filter and the DMM input impedance, which could be noticeable at the 16-bit level depending on DMM input impedance (10 MOhm vs 47 ohm = 4 LSB, for example).  If you have a >1 GOhm input impedance DMM then this is the simplest way to go, though, and you don't need the test points.  

What about using Hi-Z opamps + opto MOS switches? Of course this would add some capacitance and offset (a few tens of uV).
They could be cheaper than relays and would have one output for DMM

>> @dhslichter there are also OPTO-MOS switches that have very low charge injection due to lack of steering circuit connection. It's a LED + solar cell + FET. 1.2$/pc

>If an OPTO-MOS switch has low enough charge injection at the level we care about (we need charge injection to be ~10s of pC maximum, ideally less, over the entire range of source and drain voltages of the switch), it could work.  This is not specified in the datasheet and so would have to be measured, though (simple measurement).  We also want the resistance to be low (<1 ohm is fine) e.g. AQY211EH.  We also want the on-state resistance to be independent of drain and source voltages; I suppose you get this with an opto-isolated relay but not with a typical semiconductor analog switch.  

they are 2-wire devices. They don't have a reference to GND, these are very low capacitances of optical bareer
The question is if leakage of disabled switch is acceptable

>> The connector is a high-density 26pin DSUB. Any preference here?

>Mostly it comes down to whether there are good, well-constructed shielded cable assemblies available.  Have you had good luck finding these with this type of connector?

You need to make the cable yourself. There are metal shields for DSUB connectors easily available.
We can also use fully shielded cables like SAS, but manual assembly could be problematic

>> These relays are smaller and cheaper. And fit nicely

>Yes, those were some of the ones I was looking at.  We will dissipate 800 mW steady state just running them, but if the rest of the board is already dissipating 3.3 W steady state maybe this isn't the end of the world.  These are slightly smaller footprint than the OPTO-MOS but it's similar.  

Unfortunately they are not available as NC versions

dhslichter, 2020-10-22:
> What about using Hi-Z opamps + opto MOS switches? Of course this would add some capacitance and offset (a few tens of uV). They could be cheaper than relays and would have one output for DMM

If I understand correctly, this means putting on another set of 16 opto MOS switches -- one set to allow the output to be disconnected from the op amp (so that the voltages can be changed around for calibration), and another set to allow the op amp outputs to be routed to the single DMM connector.  Is that right?  Do we have enough space for that?

> they are 2-wire devices. They don't have a reference to GND, these are very low capacitances of optical bareer
The question is if leakage of disabled switch is acceptable

Assume that calibration of a single channel requires 100 ms.  The capacitance on the trap filters is typically something like 1 nF.  We would like to keep voltage changes to ~10s of mV at the worst during the calibration.  So this means that we can tolerate something like 100 pA of leakage current at voltage differentials up to ~25 V (max output swing of amps).  There is some wiggle room here potentially but that's the level of leakage current we are looking for.  

> You need to make the cable yourself. There are metal shields for DSUB connectors easily available.  We can also use fully shielded cables like SAS, but manual assembly could be problematic

I have a strong preference for being able to use COTS cables (you can always make custom if you want it, but don't want to be forced to), so regular 25-pin D-sub would be preferable.

> Unfortunately they are not available as NC versions

Yeah I looked for that too.

gkasprow, 2020-10-22:
> If I understand correctly, this means putting on another set of 16 opto MOS switches -- one set to allow the output to be disconnected from the op amp (so that the voltages can be changed around for [...]

no, just 16 buffering opamps connected to the outputs + optoMOS or mux (we don't care about charge injection here)
It seems relays would be cheaper anyway. We will have calibration ADC anyway, so DMM wouldn't be used everyday probably.
DMMS have Gig OHMs in 200mV or 2V range, higher ranges use dividers and offer lower impedance

> The question is if leakage of disabled switch is acceptable
>
> Assume that calibration of a single channel requires 100 ms.  The capacitance on the trap filters is typically something like 1 nF.  We [...]

we should fit within 100pA
[image]

but the capacitance varies significantly
[image]

> I have a strong preference for being able to use COTS cables (you can always make custom if you want it, but don't want to be forced to), so regular 25-pin D-sub would be preferable.

Ok, we have a place for regular D-sub. There are cables for High density versions, but hot shielded.

dhslichter, 2020-10-22:
> no, just 16 buffering opamps connected to the outputs + optoMOS or mux (we don't care about charge injection here)
It seems relays would be cheaper anyway. We will have calibration ADC anyway, so DMM wouldn't be used everyday probably.
DMMS have Gig OHMs in 200mV or 2V range, higher ranges use dividers and offer lower impedance

The purpose of this is DMM connection would be to calibrate the ADC once (not something that needs to happen often).  The ADC is already pretty good right out of the box with self-calibration, with 0.05% gain error + 1.5 mV offset error, and this calibration is to make those numbers even better if desired.  You need a nice DMM to do substantially better.  The ADC has calibration registers which you would then program on power-up with these values.  These should be pretty static in time, so you just figure out what the values should be once (i.e. when testing the board), and save them/ship them in the test report so they can be re-programmed in at power-up each time.  

If we have enough space for another 16 opto-mos relays, then we can use them to connect the output channels to a single BNC or SMA output connector from right after the buffer, but then we also need to have some digital lines to control which channel gets sent to that output connector.  Since this is generally only a one-time "factory"-type calibration, my idea was that one could use test points on the board and keep things simpler.  Such test points would make it possible for someone to redo the calibration in their lab if they want to.  But I don't think we need to have a port for actively monitoring the voltage output lines.  If people want to do that, they can build some additional inline adapter board that plugs into the D-sub output connector, etc.  I think this is not a typical use case, as long as we have a calibrated ADC on board.  

> we should fit within 100pA

yes, it seems to be good on that front

> but the capacitance varies significantly

after reading the data sheet, though, I think this refers to the capacitance between the input and output pins, so really it is only important because it presents a potential path for signals to couple even when the switch is nominally open.  For these small capacitances, I don't think we would expect much feedthrough, because when the switch is open we would only have static voltages on because we are calibrating the DAC on that channel.  So I think this is OK.  What would be a problem would be capacitance to ground that was voltage-dependent.  This means that we would get voltage-dependent rf feedthrough if we were driving a sine wave with the DACs, which is not an issue for us.  

Looks like AQY211EHAX is what we want -- if we find that it somehow has issues with the voltage swings (only can take 30 V across the switch when open) we can go to the AQY212EHAX, which has slightly higher resistance but can tolerate 60 V.

> Ok, we have a place for regular D-sub. 

I also think this just makes it a little less exotic (even though I agree D-subs are large and clunky, but they are not larger or clunkier than this box right now).

gkasprow, 2020-10-22:
So, if it is only "factory" calibration, why do we need the relays?
Or, do you plan to use relays to cancel the DAC noise during experiments?

dhslichter, 2020-10-22:
> So, if it is only "factory" calibration, why do we need the relays?
> Or, do you plan to use relays to cancel the DAC noise during experiments?

Sorry for the confusion.  There are two different calibrations that occur.  One is a calibration of the ADC, which is only a "factory" calibration.  The ADC has very low thermal drifts and so it should be pretty steady.  It is certainly much steadier (based on the spec) than the DACs and amplifier gain resistors.  

The second calibration is a regular re-calibration of the gain and offset of the each of the DAC channels, which can come from drifts in the DACs on the FMC, the differential buffer amplifier stages on the FMC, and/or the AFE amplifier stages.  The goal here is to use the ability to perform somewhat regular automated recalibrations to compensate for these drifts in the digital waveforms sent to the DACs.  Even though the DACs are only 14 bits, we are planning to use dithering/sigma-delta modulation to achieve an effective 16+ bits at DC (at NIST we have measured that we can get 18 bits, monotonic, at DC in this way with an eval board).  Such re-calibrations cannot be performed when the DAC channels are connected to the ion trap electrodes, because you have to change the DAC voltage by a lot and this will probably make the ion(s) leave the trap 😁 So ideally you would have some relay that allows you to disconnect a DAC channel from the trap briefly (the capacitance of the filters on the trap, after the AFE board, act like a sample-and-hold and maintain the voltage, as long as we keep leakage current low), so you can use the ADC to do a calibration of the DAC channel gain and offset.  Then you set the DAC back to its correct value, reconnect to the trap, all set.  Repeat for the other channels.  Then hopefully this calibration will be steady for some time until it needs to be redone (thermal drifts, etc).  

I do NOT think we need relays for the ADC calibration -- this could just be test points (in theory), although for an automated test jig you might want to use some other system (relays, or a test board that plugs into the output connector, or...).  We definitely WILL need relays for the DAC calibration, because that has to happen in an automated way while hooked up to an ion trap (you pause your quantum computing experiments to recalibrate the DACs).

dhslichter, 2020-10-22:
If the ADC is connected to the output buffers at all times, then you can also just perform periodic checking of the output voltages, either for debugging purposes or to know when a particular DAC channel seems to have drifted too far out of calibration.  Then you can decide if you need to do a full gain/offset recalibration of that channel.

gkasprow, 2020-10-23:
Thanks, now it's clear to me. We can use dual foorptints with both reed relays and OPTO-MOS with the latter ones installed.
Remember that ADCs have a sampling circuit that causes quite a high charge injection. I used to observe a few hundred mV sampling pulses on 1k input series resistors. That's why you need a low impedance ADC driver or when sampling rate is low, a capacitor which serves as a charge buffer. So I would make two copies of the output filters  - one for traps another for ADC

dnadlinger, 2020-10-23:
Wouldn't you rather want to have a slow, DC-accurate op-amp tapping off the single-ended output to the trap, so you can calibrate out any offsets from the fast op-amps?

gkasprow, 2020-10-23:
I mean a copy of the lowpass filter so the sampling current pulses produced by ADC do not interfere with trap electrodes.
To buffer with opamps, we would need 16 low offset amps.

dhslichter, 2020-10-23:
It looks like the ADC has integrated input buffers already.  There might be some charge injection due to the multiplexer switch?  However, it seems to me that if we connect this ADC to measure between IC3A and R4, and had a fairly aggressive (~10 kHz) low-pass RC with large capacitance between that node and the ADC input (using a fairly large C), it could help to smooth any glitch that comes from the ADC and reduce any impact on the trap electrodes.  

The challenge here is that the finite ~Mohm input impedance of the ADC will give a voltage divider effect if we use an RC filter here.  A low-offset buffer would do the trick, but more parts (and expensive usually).  

This voltage divider effect could just be considered to be part of the system calibration (the "factory" calibration of the ADC), as long as its tempco is sufficiently small (which would have to be tested).

[image]

gkasprow, 2020-10-23:
True. Sigma-delta converters don't have a classical S&H circuit with switched capacitances. It uses an input integrator circuit. That's why it consumes 36uA per volt without buffers. In our case, it has also a buffer. I think the charge injection from the input mux is very small and additionally divided by the input resistive network. Anyway, we don't have to use it at the moment of the experiment.

dhslichter, 2020-10-23:
so to summarize: no need for extra filters or buffers beyond what is shown on the schematic already.  Dual footprint to allow for use of opto-mos or reed relays.  Right?  Plan to populate opto-mos initially, can fall back on reed relays if necessary.

gkasprow, 2020-10-23:
right.

gkasprow, 2020-11-12:
@dhslichter what's the point in using optocouplers for digital control data? we have the same return channel anyway. We may not use these lines during experiments.

gkasprow, 2020-11-12:
I found much better relay: [AQY221N3M](http://www.farnell.com/datasheets/2872477.pdf)

dnadlinger, 2020-11-12:
There is a whole series of these small optical MOS relays, with different R*C figures-of-merit, from Panasonic and Omron.

dnadlinger, 2020-11-12:
(Panasonic makes some multi-channel ones too, AQS221…2S. For Omron, part numbers to look at for low C x R are G3VM-41PR…, but it's not clear to me that this is very relevant here.)

gkasprow, 2020-11-12:
These are already so tiny, that multi-channel versions won't save much area 👍

gkasprow, 2020-11-12:
We will fit easily with 4layer PCB and  components on on-side
[image]

gkasprow, 2020-11-12:
@dhslichter I finished the schematics of 3 boards.
Can you make a review, please?

closed 2021-02-23 by gkasprow