sinara-hw/FMC_Shuttler #15: clock distribution

gkasprow, 2020-06-14:
I plan to use clock distribution IC with LVCMOS outputs.
Clock input would be selectable: 
- Carrier FMC CLK3 BIDIR, where DRTIO recovered clock is present
- 125MHz on board generator

gkasprow, 2020-06-14:
Clock would be delivered to the following circuits:
- 8 x to DACs
- 4 x to FPGAs (2x LA, HA, HB)
- 1 x FMC CLK1 to enable further distribution in multi-board scenario

gkasprow, 2020-06-14:
There are several clock ICs that fit requirements. Examples are:
- [LMK03806](https://www.ti.com/lit/ds/symlink/lmk03806.pdf?ts=1592158596762&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FLMK03806)
- [AD9516 ](https://www.analog.com/media/en/technical-documentation/data-sheets/AD9516-0.pdf)

gkasprow, 2020-06-14:
the ICs above won't work. We need 1.8V LVCMOS.
The [ADCLK854](https://pl.mouser.com/datasheet/2/609/ADCLK854-1502771.pdf) seem to fit better

gkasprow, 2020-06-14:
I can add Si549 oscillator as the on-board clock source

gkasprow, 2020-06-14:
the ADCLK854 can be confgured as 16 LVCMOS outputs + 4 LVDS outputs. We need LVDS to route to FMC CLK inputs

dhslichter, 2020-06-15:
I like this part (especially because there are no registers to program!)  For the clocking, we need to ensure that the FPGA carrier board has the method to properly recover a high-quality clock from DRTIO (you are doing this already for other projects, right?), because this is the most likely thing that people will do.  Si549 is useful for debug.

gkasprow, 2020-11-06:
the new revision of AFC and DIOT carrier have already DRTIO clock recovery. Moreover, we can use SI549 as a part of digital PLL. External clock is also possible.

closed 2020-11-06 by gkasprow