sinara-hw/FMC_Shuttler #17: Schematic review: AFE

dhslichter, 2020-11-13:
Overall I think this looks very nice @gkasprow!  Various thoughts below:
- [ ] R11, R13 wrong values (A through H)
- [ ] would be good to do SPICE analysis for amplifier noise given resistor values (I can't do all the noise sums in my head).  Optimum source impedance seems to be ~12 kOhm given input voltage and current noise spec.
- [ ] Is there any value in splitting the RC terminations (R32, C34; R46, C35) into either R-C-R or C-R-C so they are symmetric with respect to the signal lines?  
- [ ] with no voltage regulation on board we are potentially setting ourselves up for large crosstalk because the PSRR of the op-amps is not great above ~100 kHz (50 dB @ 1 MHz, 30 dB @ 10 MHz).  With op-amps potentially driving even 10s of mA of current when running oscillating drives at a few MHz, I don't know if the on-board capacitance is enough to keep the rails from fluctuating substantially.  It also leaves us more open to switch-mode noise that gets on the rails between the power supply and the AFE box.  I would strongly prefer to have
- [ ] LM6172 specs output swings in worst case of +/-12.5 V from +/- 15 V rails (lower for higher current draw, if we are sending MHz signals through).  If we give +/- 12 V rails, it's not clear that we will be able to make the full swing to +/- 10 V for the output.  Might make sense to use +/- 15 V in, with onboard high PSRR @ MHz frequencies LDO to take that down to +/- 13 V or so?  
- [ ] P5V0 rail will get a lot of garbage when the relays switch, any concerns there?
- [ ] AVDD on the AD4115 needs to be 5 V, per data sheet, not 3.3 V.  This 5 V should not be the same rail as the relays (currently P5V0) because that is a dirty rail.
- [ ] there are 16 relay channels but only 8 indicator LEDs?

gkasprow, 2020-11-13:
I added an LDO for ADC 5V generation. It consumes 10mA of the current total (analog + digital) so LDO will dissipate roughly 80mW

gkasprow, 2020-11-13:
16 LEDs will fit 
[image]

gkasprow, 2020-11-13:
I added 3.3V LDO for LEDs and logic. It's derived from 5V rail.
5V rail is used only for relays and LEDs.

gkasprow, 2020-11-13:
Let's add LT3045 and LT3094 LDOs generating +/- 13V out of +/- 14..15V

gkasprow, 2020-11-13:
[image]

gkasprow, 2020-11-13:
I don't think that for this frequency range there would be any impact of split termination. But, with the resistor removed, there will be impedance imbalance. One needs to remember to remove or mount both components. I have a place on the PCB so I will add another capacitor to make sure we have a fully balanced pair.

dhslichter, 2020-11-13:
With LDOs added, do we want to remove the individual ferrites for each op amp?  The ferrites on the LDO inputs should help with any switch-mode spikes from the supplies, but then if we have them between the LDO and the op amps then the LDO can't do as good a job regulating the supplies for higher-frequency signals?  Recall that we may have signals of 10-15 MHz even being generated for some applications.

dnadlinger, 2020-11-13:
> […] then the LDO can't do as good a job regulating the supplies for higher-frequency signals?

I'd expect any current draw above about 1 MHz to be mainly sourced from decoupling caps, even with the new breed of very nimble LDOs. The low-pass filters formed from ferrites and decoupling caps might help localise any effects from insufficient capacitance, rather than causing more complex crosstalk issues.

gkasprow, 2020-11-13:
Yes, their role is crosstalk mitigation rather than RF isolation.
For low frequency, they are nearly pure resistance.

closed 2021-05-22 by gkasprow