sinara-hw/FMC_Shuttler #18: Schematic review: FMC card

dhslichter, 2020-11-17:
Again, looks good overall @gkasprow.  Comments and questions:

- [ ] FSADJ resistors should be chosen for 20 mA FS current instead of 10 mA.  This improves noise performance.  Value of R3, R4 should then be 2 kOhm for 1.25 V reference.
- [ ] Note on schematic that FSADJ resistors (R3, R4) should be low tempco (10 ppm/C is good).  
- [ ] DVDD (pin 7) on U1 not connected to power rail.  Should be tied to DVDDIO
- [ ] supplies don't appear on schematic (connections to PSU mezzanine missing)
- [ ] adjust amplifier resistors R34, R35, R37, R38 to account for increased I_FS (divide all values by 2)
- [ ] include pads for capacitors in parallel with R35 and R38 to allow single-pole low-pass if desired.  
- [ ] SPICE simulations for output noise of amplifiers on DAC AFE, in particular to include amp noise and effects of resistor values.  AD9117 has ~38 pA/rtHz current noise spec, about 3.5x the current noise of the amp, so this likely changes desired resistor values.
- [ ] for discussion - series pads directly before IC2 pins 3 and 5 to allow for bias current correction resistors?
- [ ] reduce common-mode voltage for AFE output (currently 1.25 V; can we make this equal to R34*10 mA~0.5 V?  This would mean this value would also be the CM voltage for AOUT_P and AOUT_N).  Using REF0V625 would be fine for this.
- [ ] note on schematic to match resistors R34 and R37, R35 and R38 (neighboring positions on tape?), also low tempco for all four (25 ppm/C or better ratio).  

More to come tomorrow, will edit top post to add.

gkasprow, 2020-11-18:
The FSADJ are already 10ppm. I replaced with 2k0

gkasprow, 2020-11-18:
DVDD is not connected intentionally. I drive DVDDIO from VADJ to make sure it works with 1.8 and 2.5V FMC signaling.

gkasprow, 2020-11-18:
Indeed, there is a missing LC filter between VADJ and VADJ_BUF

gkasprow, 2020-11-18:
Won't the bias currents of both amps compensate each other?

gkasprow, 2020-11-18:
all DACs and their clocks are within the same FPGA banks. FMC says which clocks are for which data lines. Some carriers split LA lines between 2 banks - one must make sure the right clocks are used to clock these data

gkasprow, 2020-11-18:
R34, R38 = 249R; R34, R37 = 105R

gkasprow, 2020-11-18:
We are already using REF_0V625 for AFE :) I will rename the REF signal to AFE_REF to avoid confusion

gkasprow, 2020-11-18:
I removed the voltage translation chips between FPGA and DAC because the DAC already supports the 1.8-3.3V operation (VADJ range)  and  the buffer skew reduces the timing budget

dhslichter, 2020-11-18:
> DVDD is not connected intentionally. I drive DVDDIO from VADJ to make sure it works with 1.8 and 2.5V FMC signaling.

OK, I got confused there.

> Won't the bias currents of both amps compensate each other?

They are not super well matched: -11 uA typ, ranges between -17 uA and -4 uA.  My suggestion was primarily about getting pads in place (can populate with 0R) to allow this to be changed later if desirable.  

> all DACs and their clocks are within the same FPGA banks. FMC says which clocks are for which data lines. Some carriers split LA lines between 2 banks - one must make sure the right clocks are used to clock these data

My question is -- is this a function of the specific carrier board, or is it specified in VITA 57.1?  My worry is that different FMC carrier cards might implement differently, and now you have some DACs/clocks split across banks.

dhslichter, 2020-11-18:
Still haven't completed all aspects of schematic review, will try to get it done tonight if possible.

gkasprow, 2020-11-18:
Yes, clock/banks arrangement is defined by VITA specification

gkasprow, 2020-11-18:
I routed the PCB - it still needs cosmetics, DRC and SI/PI checking

gkasprow, 2020-11-18:
the pdf is in releases

gkasprow, 2020-11-19:
the noise analysis
[image]

dhslichter, 2020-11-24:
Some questions -- the current source I1 is not the right value (should be 10 mA), also the DAC can be modeled as a current noise source with 38 pA/rtHz (datasheet).  Not sure where the 1/f knee is.  And the output noise shown here is differential noise, referred to the output?  

I think that the values of R1 and R3 should be 50 ohm, not 100 ohm?

filipswit, 2020-11-29:
- [ ] ADCLK854: Outputs 0 to 3 are set to be LVDS outputs and 4 to 11 are CMOS, so DAC_CLK7 shouldn't use 3rd output.

gkasprow, 2021-01-24:
the opamp has a drift of 3nA/K and max input current of -11...-17uA.
If we don't compensate it, the input offset value would be -600uV which is much higher than the typical Vos of 28uV
The bias currents will compensate to some extent, but if one of the inputs has -11uA and another -17uA, the imbalance would be still 6uA, which will end up with a much higher offset than Vos.
So let's add resistors at the positive input of opamps. I have to fit them somehow.

OK, so I added the noise source of the DAC. I added the FB capacitors as well.
[image]

gkasprow, 2021-01-24:
after adding bias current compensation resistors
[image]

closed 2021-02-23 by gkasprow