sinara-hw/FMC_Shuttler #22: FMC schematic: AD9117 DVDD output shorted do DVDDIO

mkiepiela, 2021-04-15:
DVDD (internal LDO output) must be disconnected from DVDDIO when supplied from external source (V_ADJ).
[image]

Disconnect and bypass DVDD output with 1 uF to GND.

[image]

dhslichter, 2021-04-15:
Do we need to maintain the potential for adjustability of the DVDDIO rail?  At least in my view, it is preferable to always require that the DVDDIO rail be at 1.8 V.  This lowers power dissipation on the DAC as well as the FPGA driving the digital lines, and running 1.8V signals instead of 3.3V signals reduces EMI.

gkasprow, 2021-05-22:
the LVDS won't work with 3.3V anyway. So we should support 1.8 and 2.5V only and this is the way I design all FMC boards. We should add 0R + 1uF to support 2.5V VADJ. We have a place for PCB for that.

gkasprow, 2021-05-22:
[image]

closed 2021-05-22 by gkasprow