sinara-hw/FMC_Shuttler #26: General Status Update

adbr8389, 2022-01-11:
@gkasprow @tprzywoz we (Ions storage) are curious if there any updates regarding the status of the shuttler design

gkasprow, 2022-01-12:
That's a question to @kaolpr. He produces it currently.

kaolpr, 2022-01-12:
We're expecting to have prototypes in ca 6 weeks. After that we'll need some time (+ca 4 weeks) to test and verify hardware and firmware.

adbr8389, 2022-01-12:
Great, thanks for the response.

adbr8389, 2023-03-16:
@kaolpr I am getting some impression of the status of the shuttler prototyping from the issues - seems like progress is being made but things are still firmly in the prototype and debug stage. Could you quickly give me a (short) coherent story of what is happening on your end? Thanks!

kaolpr, 2023-03-17:
Well, we do have v1.0 up and running. At TechnoSystem we used Digilent Genesys 2 with some minimal sample lookup table based firmware to evaluate the output signal. 

Some hardware issues were identified and already fixed for v1.1 (https://github.com/sinara-hw/Shuttler/issues/30). Beside that, an issue with bandwidth was identified (https://github.com/sinara-hw/Shuttler/issues/36). Hope to have a decision on fix / nofix soon (you're welcome to contribute). 

We did not test AFE at TechnoSystem yet.

Ongoing efforts:

- AFE testing + resolving encountered issues
- finalize v1.1
- manufacture v1.1.

We're hoping to start v1.1 manufacturing in ca 3 weeks from now.

closed 2023-11-30 by maciejprzybysz