sinara-hw/FMC_Shuttler #28: DAC_AFE-resistances

giozara, 2022-02-02:
In the DAC_AFE schematics the resistances on the ADA4896 are different from the one in the note. Is that intentional or is there a reason which eludes me?

giozara, 2022-02-02:
@gkasprow @kaolpr

gkasprow, 2022-02-02:
It's just an opamp. We need resistances that give required gain. I don't see any reason to use the same as in the datasheet

giozara, 2022-02-02:
I was just wondering, because the note in the same schematic specifies higher values which would lead to a higher gain than the one currently chosen. Were the previous (higher gain resistance pair) railing it?

gkasprow, 2022-02-03:
It's possible there is some error. Did you calculate the entire gain budget from the DAC up to the AFE output?
Here are my simulations and calculations
[SIM_CALC.zip](https://github.com/sinara-hw/Shuttler/files/7995421/SIM_CALC.zip)