sinara-hw/FMC_Shuttler #30: HW issues

gkasprow, 2022-08-04:
Missing fiducials on PSU board

gkasprow, 2022-08-04:
- [ ] check PGFB on AFE board and correct resistors

kaolpr, 2023-03-20:
In current revision (`v1.1rc1`) there is an assembly variant to connect `OSC1` either by `VADJ_BUF`  or `P1V8A`. Following note on the top sheet, the only supported `VADJ` is 1.8V. If this is our design assumption, which AFAIK is a completly valid assumption when designing FMC module, what for is that assembly variant included?

kaolpr, 2023-03-21:
> OSC1 and IC3 - Si53340 with VDD=1.8V is not capable to receive LVPECL signal (according to DS). VDD should be connected to 2.5 V or R12 R13 (LVPECL resistors) should be removed.

It is marked as done, but R12 and R13 are marked DNP only in variant. If we assume that VADJ is always 1.8V then those can be removed instead of DNP.

closed 2024-03-14 by maciejprzybysz