sinara-hw/FMC_Shuttler #32: FMC VIO_B_M2C not connected

kaolpr, 2022-10-04:
It looks like a wider problem with our open-source FMC ecosystem. `VIO_B_M2C` should be provided by the mezzanine and be used to supply IO for HBxx signals. See VITA57 rule 6.8: *"The carrier card shall utilize VIO_B_M2C as the supply voltage to the FPGA to provide the signaling standards on HBxx_y signals."*

Now, Shuttler leaves those lines floating.

Most of OHWR-related carriers just ignore those lines (AFC3, AFC4, AFCK, AFCKU) and supply HB IO from VADJ (or fixed 1.8V). DIOT FMC Carrier do have an assembly option to choose between `VIO_B_IO` and fixed 1.8V, but EEM FMC Carrier does not.

Everything is fine as long as you use affected FMC with one of those carriers. However, trying to use it with some FMC-complient carrier (e.g. Digilent Genesys2 orTrenz TEF1001),  results in unpowered (or rather partially powered via signal lines) FPGA bank.

Also trying to use FMC-complient FMC with one of those carriers may lead to having incompatibility of the electric levels.

This can be easily solved with passing through `VADJ` to `VIO_B_M2C` as an assembly variant.

closed 2024-03-14 by maciejprzybysz

gkasprow, 2024-03-18:
shouldn't we set R12 as DNP?

maciejprzybysz, 2024-04-03:
Updated in v1.2rc2