sinara-hw/FMC_Shuttler #35: AFE AUX PSU connection crossed

kaolpr, 2023-02-23:
Shuttler AFE was supposed to be compatible with AUX PSU:
[image]
However, this would require crossed cable (1-3 crossed):
[image]

gkasprow, 2023-02-23:
We would make cables anyway. They are usually available with a plug on one side.
But it's good to fix it in new revision

kaolpr, 2023-02-23:
Can we still make it to v1.1?

gkasprow, 2023-02-23:
yes
Did we produce the v1.0 prototypes?

kaolpr, 2023-02-23:
Yes. We have one.

gkasprow, 2023-02-27:
fixed
[image]

maciejprzybysz, 2024-01-22:
In v1.1 the pinout is still incorrect.

[image]

maciejprzybysz, 2024-03-14:
AUX_PSU would be updated, Shuttler AFE will be not changed.

closed 2024-03-14 by maciejprzybysz