sinara-hw/FMC_Shuttler #36: Shuttler analog bandwidth

jmatyas, 2023-02-28:
According to wiki, Shuttler's analog bandwidth is supposed to be around ~15 MHz, however after some testing on Shuttler v1.0 it seems to be closer to 6 MHz :

[image]

To achieve those results:

- a sinusoidal signal with a given frequency was generated on DAC channel
- peak-peak amplitude on one of the DAC channel's analog output (after AFE) was measured
- signal frequency was changed and the measurement was repeated

kaolpr, 2023-03-20:
@dhslichter @adbr8389 @gkasprow How do you find current bandwidth? Do we need to fix AFE to follow statement in Wiki (~15 MHz)?

kaolpr, 2023-03-20:
AFAIK this was measured before AFE. Tomorrow I'll ask @jmatyas to provide more details.

kaolpr, 2023-03-20:
But the main question is - was this intended? The simulation @gkasprow provided in #28 seems to confirm that DAC_AFE (the one on FMC) was designed for ~6 MHz.

dhslichter, 2023-03-20:
the op amps have 100 MHz GBW so I thought we would be able to get more than this? However, if we can't go any higher than what is shown here, that is probably still acceptable.

adbr8389, 2023-03-23:
What is the IC used for the differential amp after the ADA4896's? If it has some small capacitive input, I think it could explain the small discrepancy between Daniel's estimation of a 33 pF cap for a 15 MHz bandwidth.

dhslichter, 2023-03-23:
The 33 pF = 15 MHz was only a coarse estimate, I think this seems reasonable. Glad to see that this increased the bandwidth @kaolpr. Next question is whether we should increase further (e.g. 10 pF, or just DNP this capacitor). I would be in favor of testing with capacitors removed and ensuring the op amp outputs are stable (they should be, since we are at G=5). This gives maximum bandwidth out of the FMC card. One can always use following components to reduce the bandwidth as desired for applications, but this would make the FMC card maximally useful for different tasks.

closed 2024-03-14 by maciejprzybysz