sinara-hw/FMC_Shuttler #38: AFE Demultiplexer

andrewvh4, 2023-08-08:
Feature request:

Considering the proposal outlined in: [Malinowski et al, 2023: How to wire a 1000-qubit trapped ion quantum computer](https://arxiv.org/pdf/2305.12773.pdf)

An AFE capable of output demultiplexing might be useful for large-scale trapped ion systems. Has there been any discussion around this idea as a future Shuttler AFE?

gkasprow, 2023-08-09:
this is exactly what we are doing at the WUT right now
https://github.com/sinara-hw/meta/issues/46

dhslichter, 2023-08-22:
@andrewvh4 It would be possible to make a switching-based AFE of the type you mention, but this is a major technical undertaking to figure out exactly how it should work, and there is not a large demand for it right now. If your group is interested in pursuing this, you should come up with a basic plan and post it, and then people can comment. What is your specific desired use case?

Shuttler is targeting applications where >1 MHz bandwidth is needed (for diabatic transport operations, or for parametric modulation of trapping potentials). For slower operations, Zotino and Fastino have higher channel densities and are more space/cost efficient.

maciejprzybysz, 2023-11-30:
As there was no follow-up from @andrewvh4 I closed this issue. Feel free to reopen in case of any update in this topic.