sinara-hw/FMC_Shuttler #42: New revision of Shuttler

maciejprzybysz, 2024-01-22:
We are going to make new revision of Shuttler. Issues to fix:

#41 
#37 
#32 
#30 (remove R12/R13)
#27 
#35 

TBC:

#31 
#36  - I have no Shuttler hardware to test, so the safe way is to use 33pF caps -> ~12MHz BW. If someone could test 10pF or DNP I could change this accordingly. 

Estimated completion: 22/02/2024

**Feel free to add any suggestions/issues/ideas for new revision.**

dhslichter, 2024-02-08:
I think 33 pF caps should be fine for now. We do not have the ability to test yet but once we do, we can report back on different capacitor values.

maciejprzybysz, 2024-02-12:
As I didn't find any panel for v1.0/v1.1 we made panel layout for new revision - v1.2. Please check if it is suitable for you, any comments on panel inscriptions are welcome.

[image]

maciejprzybysz, 2024-03-14:
I published Shuttler v1.2 rc1 - It would be great to have community review before release/production.

I will wait till **22.03** for any remarks.

gkasprow, 2024-03-18:
LGTM apart from one issue

maciejprzybysz, 2024-04-03:
> LGTM apart from one issue

Updated in v1.2rc2, release will be added soon.